SENSIDL-PROJECT/SensIDL #15: Provide build process for the code generation

Taspolat, 2015-09-23:
Code generation needs to be supported by an automated build process, such as with Maven or Ant.

groenda, 2015-09-23:
What about classic Jenkins builds for the Eclipse infrastructure (plug-ins)? Within a code generator, there should not be so much that requires tools like Maven or Ant.

Taspolat, 2015-09-23:
True, Maven or Ant may be overkills. They were told as possible examples. But, we might also use Tycho (as a specification of Maven) to build our plugins and use Jenkins to build the Eclipse update site (#16) and provide continuous integration. Or, if Jenkins as suggested is capable of both functions (I think with the help of Buckminster), we can solve two issues with one approach.

SvenEckhardt, 2015-12-11:
##### Workaround for the buckminster guava dependency problem:

###### The Problem: 
Everything worked fine when testing it locally and there were dependencies to guava which could be resolved, but when buckminster tried to build the Project it could't resolve dependencies to guava. 

Extract from the Error Message:
 > 0    [main] ERROR e.compiler.batch.XtendBatchCompiler  - 
 > ERROR: 	CDTOFileGenerator.xtend - 
 > 3: com.google.common.base.Strings cannot be resolved to a type.

When there are no dependencies to guava in the code, the Xtend compiler still needs the dependency to guava to generate files (see https://bugs.eclipse.org/bugs/show_bug.cgi?id=418711) and so the projects couldn't be build. 

###### The Workaround:
When copying the guava JAR file in the project where guava is needed and referencing it as library in the classpath, buckminster seems to be able to resolve the dependency to guava and is able to build the project.

closed 2015-12-13 by Taspolat