SENSIDL-PROJECT/SensIDL #22: Upcoming communicationsprotocoll support

Taspolat, 2015-11-13:


Taspolat, 2015-11-15:
Outlook in communication protocol support:
* Not in focus is the generation of complex binding-APIs for specific communication protocols
* Goal: SensIDL should allow the description of data structures specified by the communication protocols. Therefore the constraints and guidelines of specific communication protocols should be considered.
** Though, the various (de-)coding functionality for different protocols should be generated.
*** Option "EncodingsSettings": Binary, JSON

* Next steps:
** For a chosen protocol (Bluetooth LE with GATT Profile) verify SensIDL regarding to its data structure description
** Adjust generators if necessary for the chosen communication protocol
** Code generation for our Android-App and test it with SensorTag that uses BLE for the receiver side

groenda, 2015-11-15:
Please state the 'specific communication protocols' explicitly for the ones that are part of the goal. Please list the goal first and be specific what should be achieved, e.g. what should be generated after this issue is solved. It should be specified in a way that allows validation/verification.
The next steps can be new issues associated with this one.

closed 2016-04-11 by SvenEckhardt