SENSIDL-PROJECT/SensIDL #32: Provide icons and label providers for metamodel elements.

groenda, 2015-12-02:
The icons and labels would allow a user-friendly presentation in the Outline of Eclipse. For example, an Interface could have an UML-Like icon -(O- and a label according to its name (rather than the technical ID). The technical ID could still be listed, e.g. comparable to measurements in EDP2 / SimuLizar.

SvenEckhardt, 2016-02-01:
to change icons for the metamodel there are two options:

1. change the getImage() method for all ItemProviders (de.fzi.sensidl.design.edit project in the de.fzi.sensidl.design.sensidl.dataRepresentation.provider package)
2. simply override the existing icons in the de.fzi.sensidl.design.edit project (icons/full/)

to change the label the getText() Method for all ItemProviders needs to be change

Taspolat, 2016-02-05:
Where can we find appropriate and free-to-use icons?

commit a53e43e2 (2016-07-24): closes #32, added custom icons for sensidl model
closed 2016-07-24 by max-peters