SENSIDL-PROJECT/SensIDL #40: Javadocs support

max-sch, 2016-01-18:
Javadocs need to be generated

SvenEckhardt, 2016-03-14:
JavaDoc is generated by Jenkins and should be published here: https://sdqbuild.ipd.kit.edu/jenkins/job/SensIDL/javadoc/ 

However there seems to be a configuration problem, because some pages of the JavaDoc such as this above are empty, but there are many pages that aren't empty (e.g https://sdqbuild.ipd.kit.edu/jenkins/job/SensIDL/javadoc/index-all.html). This shows that the JavaDoc is successfully generated and published. 

The problem with the empty index page is probably a configuration error as explained as in this post: http://forum.imagej.net/t/some-javadoc-resources-unavailable/581/2

> The Javadoc Plugin makes Javadoc available for browsing in Jenkins. The default rule set does not allow use of frames in pages served by Jenkins. To make this work again, the directives frame-src 'self' and child-src 'self' must be added to the CSP header. It appears Safari also requires the sandbox directive to be removed.

> default-src 'none'; img-src 'self'; style-src 'self'; child-src 'self'; frame-src

SvenEckhardt, 2016-03-17:
JavaDoc is released on the buildserver:

Here is the latest version of the Release: https://sdqweb.ipd.kit.edu/eclipse/sensidl/releases/latest/javadoc/
Here is the latest nightly Version: https://sdqweb.ipd.kit.edu/eclipse/sensidl/nightly/javadoc/

closed 2016-03-22 by Taspolat