SENSIDL-PROJECT/SensIDL #50: Generator Support for endianness and alignment

SvenEckhardt, 2016-02-01:


SvenEckhardt, 2016-02-22:
Alignment can by support like here [1]:
either by using bit operations(example for 16bit Alignment: ` (num + (15) & ~15`)) or by rounding manually to the next fitting value. 

[1] http://stackoverflow.com/questions/16796084/java-16-bit-alignment

The alignment should be used whenever the getter() methods are called.

pawelbielski, 2016-03-24:
Support for endianness in C:

In C, in contrast to Java, endianness depends on the machine architecture. That is why, a C-endianness-checking function [1] has to be implemented and than called to determine endianness. 

[1] (http://stackoverflow.com/questions/1001307/detecting-endianness-programmatically-in-a-c-program)

The following functionality has been implemented:

1. Check if little endian architecture
2. Swap between `LITTLE ENDIAN` and `BIG ENDIAN` endianness of data structure
3. Automatically detect endianness and swap between `LITTLE ENDIAN` and `BIG ENDIAN` if needed.

pawelbielski, 2016-03-24:
Requirements for alignment needs to be specified more precisely.

FabiScheytt, 2016-05-11:
[Regarding the C-Generators]
Double and float values can't be modified by bitwse oparators such as "<<" or ">>"
https://github.com/SENSIDL-PROJECT/SensIDL/commit/c8cf972e8bbfedf1e99221abd7770dbecdc367eb

pawelbielski, 2016-05-14:
Since the endianness of the memory representation is not part of the IEEE specification, the "<<" or ">>" bitwise operators swap the order of bytes appioprietly [1].

[1] [http://stackoverflow.com/questions/2945174/floating-point-endianness](url)

FabiScheytt, 2016-05-17:
While this may be true, none of the c compilers I tested allowed for any bitwise operator to modify a float value.
In the Link you provided the answer selected as best answer[[1]](http://stackoverflow.com/questions/2782725/converting-float-values-from-big-endian-to-little-endian/2782742#2782742) suggested to use the float as a char array and swap the individual bytes, which could work in our case: