SENSIDL-PROJECT/SensIDL #60: Move Marshalling Methods to the Utility Class

SvenEckhardt, 2016-03-16:


max-sch, 2016-03-24:
Java marshalling-methods was moved to the java utility-class

pawelbielski, 2016-06-19:
C Generators provide support for marshalling with JSON with the help of CCAN JSON libraries (json.c, json.h). Those libraries are free to use both in commercial and non-commercial projects and can be downloaded at any time from [ https://github.com/rustyrussell/ccan/tree/master/ccan/json](url). 

However, since for every single struct the code that that:
1. Allocates the JSON object
2. Populates it by mapping each field, one by one.
has to be generated, it is not possible to enclose all the needed marshalling methods within the one Utility file.

closed 2016-06-28 by pawelbielski