SENSIDL-PROJECT/SensIDL #63: Using more than one dataSet via uses in another dataSet

maPieper, 2016-03-24:
It is neccassary that we are able to use more than one dataSet in another dataSet.with _uses 
_
Example:
dataSet Characteristic **uses aValue, Uuid, Handle **{
**_      		Handle as UINT16 /**0xNNNN Attribute Handle: 16 bit value assigned by servers to their own attributes, to access the attributes */
      		Type_UUID as UINT64 constant value="10243"//AttributeTYpe: Uuid 
      		Permission as UINT8  /**Attribute Permissions */
      		Descriptor as UINT16 /** Descriptor_Handle as defined in the descriptor declaration */
      	}

Thank you!

SvenEckhardt, 2016-04-11:
More than one data sets can now be used. Java and Plantext generator generate the files the right way, c generator still only uses the first dataset aufter "uses".

closed 2016-04-15 by max-sch

pawelbielski, 2016-05-14:
The C generator can now hangle more than one dataset correctly. ( a few bugs have been fixed )