SENSIDL-PROJECT/SensIDL #65: Bluetooth LE with GATT Profile Adjustments

SvenEckhardt, 2016-04-11:
See #22: Adjust Generators and Model for the communication protocol.
Necessary Adjustments are:

- [ ] generate Methods stubs

- [ ] generate Lists

- [ ] use more datasets via "uses" (see #63)

- [ ] make getter and setter excludable

SvenEckhardt, 2016-04-11:
These Changes are all implemented in the Java Generator. The other generators still need to be changed

pawelbielski, 2016-05-14:
1. **generate Methods stubs:** In C the visibility of fields and methods is not defined.  There are a number of techniques that can approximate a private variable, but at the cost of the code clarity.
2. **generate Lists:** Since the C language does not provide support for dynamic lists by default, the length of the list could be defined explicitly in *.sidl  file ( static list ) or a linked list could be implemented ( dynamic list ). In the second case, several extra functions have to be implemented for manipulation of the list. The functionality for the lists should be defined in order to implementd appriopriate functions.
3. **use more datasets via "uses":** Multiple  datasets have the same functionality as in Java generators. 
4. **make getter and setter excludable:** Excludable getter/setter methods have the same functionality as in Java generators.

closed 2016-07-26 by Taspolat