SENSIDL-PROJECT/SensIDL #70: Linear mapping: No definition of second variable in generated c code struct

nathaliehipp, 2016-04-14:
typedef struct
{
		uint8_t radioCMD;
		uint8_t deviceType;
		uint8_t recordType;
		uint8_t identicator;
		int16_t voltage;
		float current;
		float power;
		int8_t cosPhi;
		**float adjusted_cosPhi;**
				
} Conductor;

max-sch, 2016-04-15:
Clould you give an example of the sensidl syntax that doesn't work? When I generate code with the following code:
...
cosPhi as INT8 in C adjusted by linear mapping [0;1] => [0;100] as FLOAT 
...

The C-struct is correctly generated with all specified information.

Thanks!

nathaliehipp, 2016-04-15:
Hi,
it does not make sense to create a additional variable in C code struct. 
The code should only contain the user-defined variable - no more.
The additional variable should be set outside the struct or only as return value of a method.

I hope it is clear now :-)

closed 2016-04-20 by max-sch