SENSIDL-PROJECT/SensIDL #72: C generators should generate two setters for adjusted variables

FabiScheytt, 2016-05-13:
The C generator currently only provides data conversion in a single **getter** method, while the Java generator builds two separate **setter** methods. One with data conversion and one without. To retain consistency the c generator should be adjusted to match the java generator.

pawelbielski, 2016-05-14:
The C generator match the Java generator when it comes to data adjustments.

FabiScheytt, 2016-05-17:
The c generator dosn't match the java generator. For example creates this dataset:
```
dataSet Example {
     	adjust as DOUBLE in Dimensionless adjusted with scaling factor : 5 and offset : 1
 }
```
in Java the following methods
```Java
public class Example {
//...
	public java.lang.Double getAdjust() {
		return this.adjust;
	}

	public void setAdjustWithDataConversion(java.lang.Double adjust) {
		try {
			final double offset = 1.0;
			final double scalingFactor = 5.0;
			
			this.adjust = (double) UmgebungskachelUtility.linearConversion(adjust, scalingFactor, offset);
		} catch (IllegalArgumentException e) {
			//Do something
		}
	}

	public void setAdjust(java.lang.Double adjust) {		
		this.adjust = adjust;
	} 
//...
}
```

while the c generator generates:

```c
double get_Example_adjust(Example* p){						
		return (p->adjust * 5.0) + 1.0;
	} 
void set_Example_adjust(Example* p, double adjust ) { p->adjust = adjust; }	
```