SENSIDL-PROJECT/SensIDL #81: IMappingContext from Vorto cannot be resolved after a fresh update

Taspolat, 2016-06-09:
After updating Vorto to the newest stable, a dependency issue arises for IMappingContext in SensIDLGenerator as an unresolved dependency. 

Version:  _Vorto Toolset All-in-One Bundle (Incubation)	0.4.0.201606070121	org.eclipse.vorto.feature.feature.group	Eclipse.org/Vorto_

Error message: _An internal error occurred during: "Generating Code using SensIDL Generator".
Unresolved compilation problem: 
	The type SensIDLGenerator must implement the inherited abstract method IVortoCodeGenerator.generate(InformationModel, InvocationContext)_

Taspolat, 2016-06-09:
Seems that IMappingContext is deprecated, at least cannot be found in the newest build.

SvenEckhardt, 2016-06-14:
6493c11 resolves this

closed 2016-06-14 by Taspolat