SENSIDL-PROJECT/SensIDL #86: java generator creates a second, empty constructor "random"

max-peters, 2016-06-15:
private boolean createEmptyConstructor = true

	def generateConstructorArgumentsIncludeusedDataSetss(DataSet d) {
		// create an ArrayList with all data that is not a constant NonMeasurementData (which will not be constructor arguments)
		var dataList = new ArrayList<Data>();
		var dataSet = d
		
		for (data : dataSet.eContents.filter(Data)) {
			if (data instanceof NonMeasurementData) {
				var nmdata = data as NonMeasurementData
				if (!nmdata.constant) {
					dataList.add(data)
				}
			} else if (!(data instanceof ListData)) {
				dataList.add(data)
			}
		}
		if (dataList.size > 0) {
			var firstElement = dataList.get(0).toTypeName + " " + GenerationUtil.toNameLower(dataList.get(0))
			dataList.remove(0)
			if(d.usedDataSets.size > 0){
				'''«firstElement»«FOR data : dataList», «data.toTypeName» «GenerationUtil.toNameLower(data)»«ENDFOR»«FOR pdataSet : d.usedDataSets», «GenerationUtil.toNameUpper(pdataSet)» «GenerationUtil.toNameLower(pdataSet)»«ENDFOR»'''
			} else {
				'''«firstElement»«FOR data : dataList», «data.toTypeName» «GenerationUtil.toNameLower(data)»«ENDFOR»'''
			}
		} else {
			createEmptyConstructor = false;
			''''''
		}
	}

createEmptyConstructor is true as long as there are some entries in this list. But if there is one data set with dataList.size == 0, createEmptyConstructor is false for every data set to follow.

SvenEckhardt, 2016-06-15:
You are right, this behavior was indeed not intended. 60db6e1 fixes this.

SvenEckhardt, 2016-06-15:
60db6e1 resulted in a Bug where the first constructor argument will be there twice. This is now fixed.

closed 2016-06-21 by SvenEckhardt