SENSIDL-PROJECT/SensIDL #87: java generator creates a constructor with parameters, that dont get used (only for datas adjusted by linear mapping with interval)

max-peters, 2016-06-15:
public TestDataSet5(java.lang.Byte value1, java.lang.Short value2, java.lang.Integer value3, java.lang.Long value4, java.lang.Byte rvalue1, java.lang.Short rvalue2, java.lang.Integer rvalue3, java.lang.Long rvalue4, java.lang.Byte sovalue1, java.lang.Short sovalue2, java.lang.Integer sovalue3, java.lang.Long sovalue4, java.lang.Byte lmvalue1, java.lang.Short lmvalue2, java.lang.Integer lmvalue3, java.lang.Long lmvalue4) {
		this.value1 = (byte) (value1 - java.lang.Byte.MAX_VALUE);
		this.value2 = (short) (value2 - java.lang.Short.MAX_VALUE);
		this.value3 = (int) (value3 - java.lang.Integer.MAX_VALUE);
		this.value4 = (long) (value4 - java.lang.Long.MAX_VALUE);
		this.rvalue1 = (byte) (rvalue1 - java.lang.Byte.MAX_VALUE);
		this.rvalue2 = (short) (rvalue2 - java.lang.Short.MAX_VALUE);
		this.rvalue3 = (int) (rvalue3 - java.lang.Integer.MAX_VALUE);
		this.rvalue4 = (long) (rvalue4 - java.lang.Long.MAX_VALUE);
		this.sovalue1 = (byte) (sovalue1 - java.lang.Byte.MAX_VALUE);
		this.sovalue2 = (short) (sovalue2 - java.lang.Short.MAX_VALUE);
		this.sovalue3 = (int) (sovalue3 - java.lang.Integer.MAX_VALUE);
		this.sovalue4 = (long) (sovalue4 - java.lang.Long.MAX_VALUE);
		//setLmvalue1(lmvalue1); // no setter was generated
		//setLmvalue2(lmvalue2); // no setter was generated
		//setLmvalue3(lmvalue3); // no setter was generated
		//setLmvalue4(lmvalue4); // no setter was generated
	}

Why is this only for datas adjusted by linear mapping with interval?
Why is this constructor still expecting this parameters, if it is not doing anything with them.

SvenEckhardt, 2016-06-22:
No setter was generated because it was declared with the grammar that there should be no setter (exclude: setter), so the setter can't be called from the constructor.

I'm not quite sure, why this is only for data adjusted by linear mapping, it may be because in a earlier version there was only one setter and so the variable couldn't be set in the constructor, but the setter needed to be called instead. However, since there are now more versions for the setter this isn't necessary anymore. I will fix this.

SvenEckhardt, 2016-06-22:
5bda25c fixes this by not calling the setter anymore but setting the value in the constructor instead.

closed 2016-07-12 by SvenEckhardt