SENSIDL-PROJECT/SensIDL #90: Dependency to Vorto in SensIDL Generators on updatesite

ChristophRathfelder, 2016-06-20:
Installing SensIDL using the release update site leads to an unresolved dependency:

Cannot complete the install because one or more required items could not be found.
  Software being installed: Sensidl SDK Feature 1.0.0.201606141523 (de.fzi.sensidl.language.feature.sdk.feature.group 1.0.0.201606141523)
  Missing requirement: SensIDLGenerator 1.0.0.201606141523 (de.fzi.sensidl.vorto.generator 1.0.0.201606141523) requires 'bundle org.eclipse.vorto.core 0.4.0' but it could not be found
  Cannot satisfy dependency:
    From: Sensidl SDK Feature 1.0.0.201606141523 (de.fzi.sensidl.language.feature.sdk.feature.group 1.0.0.201606141523)
    To: de.fzi.sensidl.vorto.generator [1.0.0.201606141523]

Maybe it is possible to move the vorto generator into a separated feature. If not at least the updatesite should be fixed to link to the vorto repository and automatically resolve this dependency

SvenEckhardt, 2016-06-21:
f7b6d92 moves the vorto generator to a separated feature.

To resolve this issue a workaround is, that Vorto needs to be installed before installing SensIDL so that the dependency can be resolved.

ChristophRathfelder, 2016-06-22:
Maybe it is possible to extend the update-site xml file to refere to to the Vorto Updatesite that is requires Stable or nightly. If I remember correct we have a similar problem with the Palladio updatesite to refer to another update site.
Having the URL to the Vorto update site in the eclipse configuration is enough to automatically resolve the dependencies. It is not necessary to install vorto manually.

SvenEckhardt, 2016-06-27:
There is the same Problem with Xtext, if there is no Update Site which provides Xtext (and Xtext isn't installed already)
```
Cannot complete the install because one or more required items could not be found.
  Software being installed: Sensidl SDK Feature 1.0.0.201606271419 (de.fzi.sensidl.language.feature.sdk.feature.group 1.0.0.201606271419)
  Missing requirement: Sensidl SDK Feature 1.0.0.201606271419 (de.fzi.sensidl.language.feature.sdk.feature.group 1.0.0.201606271419) requires 'org.eclipse.xtext 0.0.0' but it could not be found
```
This shows that this isn't a vorto specific Problem. 

The RMAP in the buckminster project is responsible for refering to third party Update Sites, it was my understanding, that this would be enough, but it seems like the third party update sites need to be included differently.

SvenEckhardt, 2016-07-12:
It seems like there is no other possible solution than the already mentioned workarounds:

_Either Vorto needs to be already installed or the URL to the update site should be at least in the Eclipse Configuration_

Vorto can be downloaded from the following update site: http://download.eclipse.org/vorto/update/milestones/

However, since the Vorto Generator has a separated Feature, SensIDL can be installed without this Problem, only if a user wants to use additionally the Vorto Generator, Vorto needs to be installed, which should be the case anyway.

closed 2016-07-12 by SvenEckhardt