SENSIDL-PROJECT/SensIDL #97: Reorganize SensIDL GitHub structure

joehen, 2017-05-02:
Usage of a more streamlined eclipse repo layout for SensIDL. Migrate documentation, examples and demos to separate repos.

joehen, 2017-05-04:
Proposed new structure for the SensIDL project:
## Top Level Projects

- SensIDL
   - bundles
  - features
  - releng
  - tests
- SensIDL - Vorto Integration
  - bundles
  - features
  - releng
  - tests

- SensIDL - Publications
- SensIDL - Demonstration
- SensIDL - Examples

Open question was whether we want to extract the SensIDL metamodel to a dedicated repository separated from the ui and the code generators. Moreover, we might use separate repos for the different use cases in the demonstration.

ChristophRathfelder, 2017-05-08:
For me the structure makes sense. I don't see the need to have seperated repositories for the demonstrators as well as the meta-model.

PR #98 (2020-05-18): Add build
merge refactored version to master, closes #97
closed 2020-05-18 by max-sch